espressif/ESP8266_RTOS_SDK #1245: Failed to compile when MODBUS is disabled in menuconfig

sefgit, 2023-07-25:
Using latest ESP8266_RTOS_SDK, I got following errors while compiling **uart_demo** with attached sdkconfig  (MODBUS disabled):

```
App "uart_echo" version: 1
...
...
CC build/freemodbus/common/esp_modbus_slave.o
In file included from E:/idf/msys32/home/user/esp/ESP8266_RTOS_SDK/components/freemodbus/port/port.h:21,
                 from E:/idf/msys32/home/user/esp/ESP8266_RTOS_SDK/components/freemodbus/modbus/include/mb.h:34,
                 from E:/idf/msys32/home/user/esp/ESP8266_RTOS_SDK/components/freemodbus/common/esp_modbus_callbacks.h:21,
[sdkconfig.txt](https://github.com/espressif/ESP8266_RTOS_SDK/files/12166234/sdkconfig.txt)

                 from E:/idf/msys32/home/user/esp/ESP8266_RTOS_SDK/components/freemodbus/common/mbc_slave.h:23,
                 from E:/idf/msys32/home/user/esp/ESP8266_RTOS_SDK/components/freemodbus/common/esp_modbus_slave.c:18:
E:/idf/msys32/home/user/esp/ESP8266_RTOS_SDK/components/freemodbus/modbus/include/mbconfig.h:66:2: error: #error "None of Modbus communication mode is enabled. Please enable one of (ASCII, RTU, TCP) mode in Kconfig."
 #error "None of Modbus communication mode is enabled. Please enable one of (ASCII, RTU, TCP) mode in Kconfig."
  ^~~~~
In file included from E:/idf/msys32/home/user/esp/ESP8266_RTOS_SDK/components/freemodbus/common/esp_modbus_callbacks.h:22,
                 from E:/idf/msys32/home/user/esp/ESP8266_RTOS_SDK/components/freemodbus/common/mbc_slave.h:23,
                 from E:/idf/msys32/home/user/esp/ESP8266_RTOS_SDK/components/freemodbus/common/esp_modbus_slave.c:18:
E:/idf/msys32/home/user/esp/ESP8266_RTOS_SDK/components/freemodbus/modbus/include/mb_m.h:401:1: error: unknown type name 'eMBMasterErrorEventType'; did you mean 'eMBEventType'?
 eMBMasterErrorEventType eMBMasterGetErrorType( void );
 ^~~~~~~~~~~~~~~~~~~~~~~
 eMBEventType
E:/idf/msys32/home/user/esp/ESP8266_RTOS_SDK/components/freemodbus/modbus/include/mb_m.h:402:29: error: unknown type name 'eMBMasterErrorEventType'; did you mean 'eMBEventType'?
 void vMBMasterSetErrorType( eMBMasterErrorEventType errorType );
                             ^~~~~~~~~~~~~~~~~~~~~~~
                             eMBEventType
make[1]: *** [/home/user/esp/ESP8266_RTOS_SDK/make/component_wrapper.mk:292: common/esp_modbus_slave.o] Error 1
make: *** [/home/user/esp/ESP8266_RTOS_SDK/make/project.mk:571: component-freemodbus-build] Error 2
```
[sdkconfig.txt](https://github.com/espressif/ESP8266_RTOS_SDK/files/12166255/sdkconfig.txt)

ioviedodev, 2024-07-12:
Hi, comunitty, I have the same issue ESP-IDF v4.4.6. Some can help us? Please

ioviedodev, 2024-07-12:
It looks like that is mandatory enable at least one of the modbus mode to compile the project enven thoug you can not need it.