espressif/ESP8266_RTOS_SDK #1251: /xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/lib/crt0.o:(.literal+0x0): undefined reference to `main' collect2: error: ld returned 1 exit status make[1]: *** [/Users/xu756/Downloads/hello_world/build/bootloader/bootloader.elf] Error 1 make: *** [/Users/xu756/Downloads/hello_world/build/bootloader/bootloader.bin] Error 2

xu756, 2023-09-11:
Have installed following [this guide](xtensa-lx106-elf/lib/crt0.o:(.literal+0x0): undefined reference to `main'). When doing make got the error

```

# xu756 @ bogon in ~/Downloads/hello_world [17:51:17] 
$ make           
Toolchain path: /Users/xu756/esp/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc
Toolchain version: esp-2020r3-49-gd5524c1
Compiler version: 8.4.0
Python requirements from /Users/xu756/esp/ESP8266_RTOS_SDK/requirements.txt are satisfied.
Building partitions from /Users/xu756/esp/ESP8266_RTOS_SDK/components/partition_table/partitions_singleapp.csv...
CC build/bootloader/main/bootloader_start.o
AR build/bootloader/main/libmain.a
CC build/bootloader/bootloader_support/src/bootloader_clock.o
CC build/bootloader/bootloader_support/src/bootloader_common.o
CC build/bootloader/bootloader_support/src/bootloader_flash.o
CC build/bootloader/bootloader_support/src/bootloader_init.o
CC build/bootloader/bootloader_support/src/bootloader_random.o
CC build/bootloader/bootloader_support/src/bootloader_sha.o
CC build/bootloader/bootloader_support/src/bootloader_utility.o
CC build/bootloader/bootloader_support/src/efuse.o
CC build/bootloader/bootloader_support/src/esp_image_format.o
CC build/bootloader/bootloader_support/src/flash_encrypt.o
CC build/bootloader/bootloader_support/src/flash_partitions.o
CC build/bootloader/bootloader_support/src/flash_qio_mode.o
CC build/bootloader/bootloader_support/src/secure_boot.o
CC build/bootloader/bootloader_support/src/secure_boot_signatures.o
AR build/bootloader/bootloader_support/libbootloader_support.a
CC build/bootloader/spi_flash/src/spi_flash.o
CC build/bootloader/spi_flash/src/spi_flash_raw.o
CC build/bootloader/spi_flash/port/port.o
AR build/bootloader/spi_flash/libspi_flash.a
CC build/bootloader/log/log.o
AR build/bootloader/log/liblog.a
CC build/bootloader/esp8266/source/ets_printf.o
CC build/bootloader/esp8266/source/crc.o
CC build/bootloader/esp8266/source/esp_fast_boot.o
AR build/bootloader/esp8266/libesp8266.a
LD build/bootloader/bootloader.elf
/Users/xu756/esp/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld: /Users/xu756/esp/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/lib/crt0.o:(.literal+0x0): undefined reference to `main'
collect2: error: ld returned 1 exit status
make[1]: *** [/Users/xu756/Downloads/hello_world/build/bootloader/bootloader.elf] Error 1
make: *** [/Users/xu756/Downloads/hello_world/build/bootloader/bootloader.bin] Error 2

# xu756 @ bogon in ~/Downloads/hello_world [17:51:24] C:2
$ 

```

macos13.5

JIMS-Innovations, 2024-06-28:
I found a solution to this by deleting the `int` before the `main` in "ESP8266_RTOS_SDK/tools/kconfig/lxdialog/check-lxdialog.sh" under "#include CURSES_LOC" or any other prefix before `main`

tikhonp, 2024-11-01:
I have the same problem. Did you figured out a solution?

I'm trying to compile it on MacOS 15.1 with Apple M2 Pro silicone. (`Darwin-arm64`)