espressif/ESP8266_RTOS_SDK #1254: requirements.txt not found

ZephyrKeXiner, 2023-09-29:
## Environment

- Development Kit:NodeMCU ESP8266
- IDF version : 015f3099
- Development Env:   Make
- Operating System:   Ubuntu
- Power Supply:  USB

## Problem Description
When I run the command 'python -m pip install --user -r $IDF_PATH/requirements.txt',the output is ''ERROR: Could not open requirements file: [Errno 2] No such a file or dictionary: '~/esp/ESP8266_RTOS_SDK/requirements.txt'.

but it's EXACTLY in my ~/esp/ESP8266_RTOS_SDK.

I guess it's because the python venv,but I can't locate the problem.

### Expected Behavior
It will work correctly
### Actual Behavior
It shows 'No such a file or dictionary',while it IS here.
### Steps to repropduce
Sorry,I don't how to reproduce.
I try to install requirements in requirements.txt,and it's happened.

## Other items if possible

- [ ] It also appears when I use clangd config command ''cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=1"

Olewojt, 2024-07-16:
It's because of home symbol used in path assigned to IDF_PATH. When creating environmental variables you should use `$HOME` which will return absolute path to your home directory e.g `/home/user`
`export IDF_PATH='$HOME/esp/ESP8266_RTOS_SDK'`

ZephyrKeXiner, 2024-07-29:
It really works!
thank you very much!

closed 2024-07-29 by ZephyrKeXiner