espressif/ESP8266_RTOS_SDK #1257: Missing option to generate and flash external spiffimg

Asanga-Viraj, 2023-10-26:
In IDF for ESP32, there is an option to generate external project folder into spiffimg.bin and flash it to ESP32 chip. Can use following line in cmakelist.txt inside the main.c folder.

`spiffs_create_partition_image(storage ../data FLASH_IN_PROJECT) `

ESP8266, this command is not working and I couldn't find any option to do it. Is there a way exists?

praveens-git, 2024-03-12:
Are there any updates on this?

AMKrol, 2024-03-13:
I don't see any similar function in SDK, but...

You can use a [mkspiffs](https://github.com/igrr/mkspiffs) project. 

1. All its need is to compile the project with the parameters:
```
make dist CPPFLAGS="-DSPIFFS_OBJ_NAME_LEN=32 -DCONFIG_SPIFFS_META_LENGTH=4 -DSPIFFS_USE_MAGIC=1 -DSPIFFS_USE_MAGIC_LENGTH=1 -DSPIFFS_ALIGNED_OBJECT_INDEX_TABLES=4" BUILD_CONFIG_NAME=-custom
```

2. Next create the partition image from files in directory (e.g. `data` folder)
```
./mkspiffs -p 256 -b 4096 -s 458752 -d 5 -c ./data ./spiffs.bin
```

3. Upload image to the ESP8266 to data storage partition:
```
esptool.py --port /dev/ttyUSB0 write_flash 0x90000 /app/spiffs.bin
```