espressif/ESP8266_RTOS_SDK #1271: Proble Install Python Package by Using mingw32 on Windows

JustAnotherHobbyist, 2024-03-25:
----------------------------- Delete above -----------------------------

## Environment

- Development Kit: ESP8266_RTOS_SDK
- Development Env:      MSYS32
- Operating System:     Windows 11 x64

## Problem Description

Problem by installing the Python Packages with the command "python -m pip install --user -r $IDF_PATH/requirements.txt" in the mingw32.exe command prompt.

### Solved  by
Add the --no-build-isolation solved this issue solved the problem:

"python -m pip install --no-build-isolation --user -r $IDF_PATH/requirements.txt"

maelsilva96, 2024-04-07:
Hi @JustAnotherHobbyist, i have same problema, change your requirements.txt and define limit max version in pyelftools to 0.28.

My file requirements.txt.

```
# This is a list of python packages needed for ESP-IDF. This file is used with pip.
# Please see the Get Started section of the ESP-IDF Programming Guide for further information.
#
setuptools
# The setuptools package is required to install source distributions and on some systems is not installed by default.
# Please keep it as the first item of this list.
#
click>=5.0
pyserial>=3.0
future>=0.15.2
cryptography>=2.1.4,<35
pyparsing>=2.0.3,<2.4.0
pyelftools>=0.22,<0.28
```

dvof, 2024-06-20:
seems like it fails at `0.31`, so ` pyelftools>=0.22,<0.31` works as well

hquiroz-1988, 2024-07-19:
Also fixed by either fixing the python packages or giving them an upper boundary, here are mine at the moment.

`# This is a list of python packages needed for ESP-IDF. This file is used with pip.
# Please see the Get Started section of the ESP-IDF Programming Guide for further information.
#
setuptools==40.4.3
# The setuptools package is required to install source distributions and on some systems is not installed by default.
# Please keep it as the first item of this list.
#
click==5.0
pyserial==3.0
future==0.15.2
cryptography==2.3.1
pyparsing==2.0.3
pyelftools==0.22`

zjt-code, 2024-12-02:
> 通过修复 python 包或为其提供上限来修复，目前这是我的。
> 
> 调整`ESP8266_RTOS_SDK/requirements.txt`
> 
> `setuptools==40.4.3` `click==5.0` `pyserial==3.0` `future==0.15.2` `cryptography==2.3.1` `pyparsing==2.0.3` `pyelftools==0.22`

It is working