espressif/ESP8266_RTOS_SDK #1274: Error in configuring ESP8266 RTOS SDK development environment on Linux？？？

UFO281, 2024-04-06:
when I run this command 

 wls@ubuntu:~/esp/ESP8266_RTOS_SDK/examples/get-started/hello_world$ make all

Encountered this error：

Building partitions from /home/wls/esp/ESP8266_RTOS_SDK/components/partition_table/partitions_singleapp.csv...
WARNING: Missing submodule components/json/cJSON...
Attempting 'git submodule update --init components/json/cJSON' in esp-idf root directory...
Submodule 'components/json/cJSON' (git@github.com:DaveGamble/cJSON.git) registered for path 'components/json/cJSON'
Cloning into '/home/wls/esp/ESP8266_RTOS_SDK/components/json/cJSON'...
Submodule path 'components/json/cJSON': checked out '3c8935676a97c7c97bf006db8312875b4f292f6c'
WARNING: Missing submodule components/lwip/lwip...
Attempting 'git submodule update --init components/lwip/lwip' in esp-idf root directory...
Submodule 'components/lwip/lwip' (git@github.com:espressif/esp-lwip.git) registered for path 'components/lwip/lwip'
Cloning into '/home/wls/esp/ESP8266_RTOS_SDK/components/lwip/lwip'...

Submodule path 'components/lwip/lwip': checked out '4fd84abace0cd8a5a709c194fa54afb3cd39b3f7'
WARNING: Missing submodule components/mbedtls/mbedtls...
Attempting 'git submodule update --init components/mbedtls/mbedtls' in esp-idf root directory...
Submodule 'components/mbedtls/mbedtls' (git@github.com:espressif/mbedtls.git) registered for path 'components/mbedtls/mbedtls'
Cloning into '/home/wls/esp/ESP8266_RTOS_SDK/components/mbedtls/mbedtls'...

UFO281, 2024-04-06:
who can help me ?????  please!!

UFO281, 2024-04-06:
The development environment of Lexin's ESP8266 is too complex!

karnigen, 2024-04-08:
For me, everything works fine for version 3.4 as it should. Just use `make menuconfig` and` make flash`, and it works right away. Maybe for ESP8266, it's not as well-documented as for ESP32, so if needed, look for inspiration there.

It seems that the RTOS for ESP8266 was not installed as described.
- Check if you really cloned the git with the recursive switch.
- Check if version v3.4 was checked out from the git.
- Check if you ran install.sh.
- Check if the compilers are stored in `$HOME/.espressif` and if a virtual Python environment has been created.
- Activate Python for ESP8266 RTOS from `$HOME/.espressif` and run export.sh.
- It's absolutely crucial for the environment to be correctly set up to avoid being overwritten by running any IDE, and to have the correct virtual Python environment active.

0smr, 2024-04-09:
@UFO281 As far as I can see, there are no errors in your logs.

UFO281, 2024-04-09:
> @UFO281 As far as I can see, there are no errors in your logs.

But he can't work properly

UFO281, 2024-04-13:
no, had error is compile error

XiaoXianNv-boot, 2024-05-07:
子模块未初始化,请检查网络