espressif/ESP8266_RTOS_SDK #1275: Segment loaded at 0x*** lands in same 64KB flash mapping as segment loaded at 0x***

0smr, 2024-04-09:
## Problem Description

I am experiencing a linker issue when trying to disable debug assertion. This also occurs when attempting to compile my application in Release/MinSizeRelease mode in Qt Creator.

- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

## Environment

- IDF version: `c4634015d873c6cf86e118df9364c2fbc20d9a11` (latest version)
- Development Env: QtCreator
- Operating System: Windows

### Steps to reproduce

1. Add `CONFIG_COMPILER_OPTIMIZATION_ASSERTIONS_DISABLE=y`
2. Compile

### Code to reproduce this issue
A simple blink.

```cpp
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"

#include "driver/gpio.h"
#include "esp8266/gpio_struct.h"
#include "esp8266/gpio_register.h"

extern "C" void app_main() {
    gpio_config_t config {
        (0x1 << 0x4),
        GPIO_MODE_OUTPUT,
        GPIO_PULLUP_DISABLE,
        GPIO_PULLDOWN_DISABLE,
        GPIO_INTR_DISABLE,
    };

    gpio_config(&config);

    while(true) {
        uint32_t lastState = GPIO.out;

        GPIO.out_w1tc |= lastState & (0x1 << 0x4);
        GPIO.out_w1ts |= ~lastState & (0x1 << 0x4);

        fflush(stdout);
        vTaskDelay(1000 / portTICK_RATE_MS); // 1 second delay
    }
}
```

## Compile Logs

```log
03:08:14: Running steps for project app...
03:08:14: Running CMake in preparation to build...
03:08:31: Starting: "C:\Program Files\CMake\bin\cmake.exe" --build C:/<path-to-build-directory>/build-app-ESP_idf_freamwork_toolchain-Debug --target all size
[1/590 0.6/sec] Building C object esp-idf/esp_event/CMakeFiles/__idf_esp_event.dir/default_event_loop.c.obj
[2/590 1.1/sec] Building C object esp-idf/esp_event/CMakeFiles/__idf_esp_event.dir/esp_event_private.c.obj
[3/590 1.6/sec] Building C object esp-idf/esp_event/CMakeFiles/__idf_esp_event.dir/event_loop_legacy.c.obj
[4/590 2.0/sec] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/aesni.c.obj
...
A lot of building and linking stuff...
...
[583/590 4.3/sec] Linking CXX static library esp-idf\mqtt\libmqtt.a
[584/590 4.3/sec] Linking CXX static library esp-idf\libsodium\liblibsodium.a
[585/590 4.3/sec] Linking CXX static library esp-idf\esp_https_ota\libesp_https_ota.a
[586/590 4.3/sec] Linking CXX static library esp-idf\main\libmain.a
[587/590 4.2/sec] Generating ld/esp8266.project.ld
[588/590 4.1/sec] Linking CXX executable app.elf
[589/590 4.1/sec] Generating binary image from built executable
FAILED: .bin_timestamp C:/<path-to-build-directory>/build-app-ESP_idf_freamwork_toolchain-Debug/.bin_timestamp 
cmd.exe /C "cd /D C:\<path-to-build-directory>\build-app-ESP_idf_freamwork_toolchain-Debug && python C:/Espressif/esp8266_rtos_sdk/components/esptool_py/esptool/esptool.py --chip esp8266 elf2image --flash_mode dio --flash_freq 26m --flash_size 4MB --version=3 -o C:/<path-to-build-directory>/build-app-ESP_idf_freamwork_toolchain-Debug/app.bin app.elf && "C:\Program Files\CMake\bin\cmake.exe" -E echo "Generated C:/<path-to-build-directory>/build-app-ESP_idf_freamwork_toolchain-Debug/app.bin" && "C:\Program Files\CMake\bin\cmake.exe" -E md5sum C:/<path-to-build-directory>/build-app-ESP_idf_freamwork_toolchain-Debug/app.bin > C:/<path-to-build-directory>/build-app-ESP_idf_freamwork_toolchain-Debug/.bin_timestamp"
esptool.py v2.4.0
.flash.rodata len 0x029f4 load 0x4022ca00

A fatal error occurred: Segment loaded at 0x4022ca00 lands in same 64KB flash mapping as segment loaded at 0x40220010. Can't generate binary. Suggest changing linker script or ELF to merge sections.
[590/590 4.0/sec] cmd.exe /C "cd /D C:\<path-to-build-directory>\build-app-ESP_idf_freamwork_toolchain-Debug && python C:/Espressif/esp8266_rtos_sdk/tools/idf_size.py C:/<path-to-build-directory>/build-app-ESP_idf_freamwork_toolchain-Debug/app.map"
Total sizes:
 DRAM .data size:     876 bytes
 DRAM .bss  size:    4856 bytes
Used static DRAM:    5732 bytes (  92572 available, 5.8% used)
Used static IRAM:   16564 bytes (  32588 available, 33.7% used)
      Flash code:   51694 bytes
    Flash rodata:   10740 bytes
Total image size:~  79874 bytes (.bin may be padded larger)
ninja: build stopped: subcommand failed.
03:10:58: The process "C:\Program Files\CMake\bin\cmake.exe" exited with code 1.
Error while building/deploying project app (kit: ESP8266 toolchain)
When executing step "Build"
03:10:58: Elapsed time: 02:44.
```

## Other stuff

### SDK config file:
```ini
# the default settings, just following change:
CONFIG_COMPILER_OPTIMIZATION_ASSERTIONS_DISABLE=y
```

[sdkconfig](https://github.com/espressif/ESP8266_RTOS_SDK/files/14912033/sdkconfig.txt)

### Objdump
```
$xtensa-lx106-elf-objdump -h app.elf

app.elf:     file format elf32-xtensa-le

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .rtc_noinit   00000000  60001200  60001200  00015aa0  2**0
                  CONTENTS
  1 .iram0.vectors 00000080  40100000  40100000  00002000  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  2 .iram0.text   00003ec8  40100080  40100080  00002080  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  3 .iram0.bss    00000208  40103f48  40103f48  00005f48  2**2
                  ALLOC
  4 .dram0.data   00000374  3ffe8000  3ffe8000  00001000  2**4
                  CONTENTS, ALLOC, LOAD, DATA
  5 .noinit       00000000  3ffe8374  3ffe8374  00015aa0  2**0
                  CONTENTS
  6 .dram0.bss    000012f8  3ffe8380  3ffe8380  00001374  2**4
                  ALLOC
  7 .flash.text   0000cfb6  40210010  40210010  00006010  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  8 .flash.rodata 00002ad8  4021cfc8  4021cfc8  00012fc8  2**2
                  CONTENTS, ALLOC, LOAD, DATA
  9 .xtensa.info  00000038  00000000  00000000  00015aa0  2**0
                  CONTENTS, READONLY
 10 .comment      0000008c  00000000  00000000  00015ad8  2**0
                  CONTENTS, READONLY
 11 .xt.prop._ZTISt9exception 0000000c  00000000  00000000  00015b64  2**0
                  CONTENTS, READONLY
 12 .xt.prop._ZTISt13bad_exception 0000000c  00000000  00000000  00015b70  2**0
                  CONTENTS, READONLY
 13 .xt.prop._ZTIN10__cxxabiv115__forced_unwindE 0000000c  00000000  00000000  00015b7c  2**0
                  CONTENTS, READONLY
 14 .xt.prop._ZTIN10__cxxabiv119__foreign_exceptionE 0000000c  00000000  00000000  00015b88  2**0
                  CONTENTS, READONLY
 15 .xt.prop._ZTVSt13bad_exception 0000000c  00000000  00000000  00015b94  2**0
                  CONTENTS, READONLY
 16 .xt.lit._ZNK9__gnu_cxx24__concurrence_lock_error4whatEv 00000008  00000000  00000000  00015ba0  2**0
                  CONTENTS, READONLY
```

Thalhammer, 2024-04-10:
I am hitting the same issue. Using Assertion level silent works fine, but as soon as assertions are disable entirely it breaks. Right now I am using silent mode, but switching to none would probably remove yet another couple bytes I can use for a feature in 1M (split into two ota parts) devices.

0smr, 2024-04-13:
@Thalhammer, thank you for your comment. I will close this issue since you answered it in #1262.

closed 2024-04-13 by 0smr