espressif/ESP8266_RTOS_SDK #1278: menuconfig script fails for ncurses in Arch-Linux

selimkeles, 2024-05-11:
Hi 

I have updated my IDF master branch to the latest version and checked that the issue is present there. I have searched the issue tracker for a similar issue and not found a similar issue.

Using Arch Linux up-to-date kernel and all dependencies are installed. Running make menuconfig command for most basic project hello world causes error which is given below. Since archlinux contains ncurses lib inside /usr/include/ I dont know why source code does not detects it place:

```bash
$ make menuconfig 
make[1]: Entering directory '/home/msk/Documents/Repos/esp/ESP8266_RTOS_SDK/tools/kconfig'
/home/msk/Documents/Repos/esp/ESP8266_RTOS_SDK/tools/kconfig/lxdialog/check-lxdialog.sh -check cc -D_DEFAULT_SOURCE -D_XOPEN_SOURCE=600 -DCURSES_LOC="<ncurses.h>" -DNCURSES_WIDECHAR=1 -DLOCALE -MMD -MP -I "." -I "/home/msk/Documents/Repos/esp/ESP8266_RTOS_SDK/tools/kconfig" -lncursesw
 *** Unable to find the ncurses libraries or the
 *** required header files.
 *** 'make menuconfig' requires the ncurses libraries.
 *** 
 *** Install ncurses (ncurses-devel) and try again.
 *** 
make[1]: *** [Makefile:219: dochecklxdialog] Error 1
make[1]: Leaving directory '/home/msk/Documents/Repos/esp/ESP8266_RTOS_SDK/tools/kconfig'
make: *** No rule to make target '/home/msk/Documents/Repos/esp/ESP8266_RTOS_SDK/tools/kconfig/conf-idf', needed by '/home/msk/Documents/Repos/esp/ESP8266_RTOS_SDK/tools/kconfig/mconf-idf'.  Stop.

```

my env variables:
```bash
$ echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/bin:/usr/lib/jvm/java-22-openjdk/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/home/msk/.espressif/tools/xtensa-lx106-elf/esp-2020r3-49-gd5524c1-8.4.0/xtensa-lx106-elf/bin

```
```bash
$ echo $IDF_PATH 
/home/msk/Documents/Repos/esp/ESP8266_RTOS_SDK
```

also I installed python3 packages for dependent libraries.

ncurses available on system as you can see:
```bash
$ sudo pacman -Q ncurses
ncurses 6.4_20230520-3
```

saharulik, 2024-05-14:
Hi, firstly I can say actually I am not pro and I am not sure why they did that, but in file  tools/kconfig/lxdialog, you can add int in `check() {
        $cc -x c - -o $tmp 2>/dev/null <<'EOF'
#include CURSES_LOC
main() {}
EOF` 
to main 
`check() {
        $cc -x c - -o $tmp 2>/dev/null <<'EOF'
#include CURSES_LOC
int main() {}
EOF` . 
I don't know firstly reason why they don't do that but it's help me

fiffy326, 2024-05-15:
If anyone uses the AUR package, I've made an updated PKGBUILD that applies the fix @saharulik mentioned to use in the meantime until the maintainer updates the package.

selimkeles, 2024-05-15:
AUR package last updated years ago. Is it normal to use it since master branch have couple of new commits even last month?

fiffy326, 2024-05-15:
The AUR package sources its files from this repo, so it is up to date with the current release.

selimkeles, 2024-05-17:
@fiffy326 thanks for warning also repo maintainer fixed issue with your warning. I'll close this issue.

closed 2024-05-17 by selimkeles