espressif/ESP8266_RTOS_SDK #1283: Invalid error caused by menuconfig: Unable to find the ncurses libraries or the required header files.

DaniilAlpha, 2024-07-09:
[Similar issue](https://github.com/espressif/ESP8266_RTOS_SDK/issues/1278) do exist, but is closed and not very relevant as about an arch package.

## Environment

- Development Kit:
- IDF version:          d3a5f99ef8acecb3e0fbba8493aba15fa50fe375
- Development Env:      CMake
- Operating System:     Fedora 40 (6.9.7-200.fc40.x86_64)

## Problem Description

After idf.py menuconfig is run, ESP8266_RTOS_SDK/tools/kconfig/lxdialog/check-lxdialog.sh reports an error
```
 *** Unable to find the ncurses libraries or the
 *** required header files.
 *** 'make menuconfig' requires the ncurses libraries.
 *** 
 *** Install ncurses (ncurses-devel) and try again.
 *** 
```
Althought, both ncurses and ncurses-devel is preset on the system. 
After a bit of digging, I realized that error is actually not related to ncurses at all. The way script checks for ncurses presence is ~~kida lame~~ not so good, furthermore it relies on non-standard C (ommiting type before the `main` function), so compiler generates a warning which makes the check fail. 

### Expected Behavior

Menuconfig builds out of the box.

### Actual Behavior

To build menuconfig, tinkering with idf scripts is required.

### Steps to repropduce

May be kind of tricky, as it depends on compiler options and the compiler itself. I personaly add additional public warnings (-Wall -Wextra -Wformat) at the root CMakeLists, but not sure if they are even got used in this case. As for the compiler, it's taken from link from [wiki](https://docs.espressif.com/projects/esp8266-rtos-sdk/en/latest/get-started/linux-setup.html).

### Fix

Adding `int` before this test `main` function fixes the problem.

mnganga10, 2024-09-02:
very helpful, should be pushed to master branch

tikhonp, 2024-11-01:
yeap this pull request fixes problem for me

romainreignier, 2025-12-17:
#1287 fixed the issue on Debian 13 for me.

alfadelta10010, 2026-02-08:
#1287 fixed the issue for me on Debian 14, many thanks