espressif/ESP8266_RTOS_SDK #1296: Clone error: 'coap/libcoap/ext/tinydtls The requested URL returned error: 502'

viordash, 2025-01-26:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Clone with submodules update will be successful.

I think [libcoap](https://github.com/obgm/libcoap) needs to be updated to [v4.2.1](https://github.com/obgm/libcoap/releases/tag/v4.2.1)

Thanks

### Actual behavior (suspected bug)

When cloning esp-idf with updating submodules, the error `Failed to clone 'ext/tinydtls`. occurs.

### Error logs or terminal output

```shell
Submodule path 'components/coap/libcoap': checked out 'cfec0d072c5b99ed3e54828ca50ea2f6b91e1f50'
Submodule 'ext/tinydtls' (https://git.eclipse.org/r/tinydtls/org.eclipse.tinydtls) registered for path 'components/coap/libcoap/ext/tinydtls'
Cloning into '/home/viordash/Downloads/ESP8266_RTOS_SDK/components/coap/libcoap/ext/tinydtls'...
fatal: unable to access 'https://git.eclipse.org/r/tinydtls/org.eclipse.tinydtls/': The requested URL returned error: 403
fatal: clone of 'https://git.eclipse.org/r/tinydtls/org.eclipse.tinydtls' into submodule path '/home/viordash/Downloads/ESP8266_RTOS_SDK/components/coap/libcoap/ext/tinydtls' failed
Failed to clone 'ext/tinydtls'. Retry scheduled
Cloning into '/home/viordash/Downloads/ESP8266_RTOS_SDK/components/coap/libcoap/ext/tinydtls'...
fatal: unable to access 'https://git.eclipse.org/r/tinydtls/org.eclipse.tinydtls/': The requested URL returned error: 403
fatal: clone of 'https://git.eclipse.org/r/tinydtls/org.eclipse.tinydtls' into submodule path '/home/viordash/Downloads/ESP8266_RTOS_SDK/components/coap/libcoap/ext/tinydtls' failed
Failed to clone 'ext/tinydtls' a second time, aborting
Submodule path 'components/json/cJSON': checked out '3c8935676a97c7c97bf006db8312875b4f292f6c'
Submodule path 'components/lwip/lwip': checked out '4fd84abace0cd8a5a709c194fa54afb3cd39b3f7'
Submodule path 'components/mbedtls/mbedtls': checked out '9ef92c551eb8d92677034c3ec8078a8076febf41'
Submodule path 'components/mqtt/esp-mqtt': checked out '01594bf118ae502b5a0ead040446f2be75d26223'
fatal: Failed to recurse into submodule path 'components/coap/libcoap'
```

### Steps to reproduce the behavior

git clone --recurse-submodules https://github.com/espressif/ESP8266_RTOS_SDK.git

### Project release version

v3.4

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Linux 64

### Shell

ZSH

NostalgistL00, 2025-03-08:
I just meet the same question,nothing works right now.

SpComb, 2025-04-19:
The error has changed slightly:

```
#10 15.28 Cloning into '/opt/ESP8266_RTOS_SDK/components/coap/libcoap/ext/tinydtls'...
#10 16.32 fatal: unable to update url base from redirection:
#10 16.32   asked for: https://git.eclipse.org/r/tinydtls/org.eclipse.tinydtls/info/refs?service=git-upload-pack
#10 16.32    redirect: https://github.com/eclipse-tinydtls
#10 16.32 fatal: clone of 'https://git.eclipse.org/r/tinydtls/org.eclipse.tinydtls' into submodule path '/opt/ESP8266_RTOS_SDK/components/coap/libcoap/ext/tinydtls' failed
#10 16.32 Failed to clone 'ext/tinydtls'. Retry scheduled
#10 16.32 Cloning into '/opt/ESP8266_RTOS_SDK/components/coap/libcoap/ext/tinydtls'...
#10 18.03 fatal: unable to update url base from redirection:
#10 18.03   asked for: https://git.eclipse.org/r/tinydtls/org.eclipse.tinydtls/info/refs?service=git-upload-pack
#10 18.03    redirect: https://github.com/eclipse-tinydtls
#10 18.04 fatal: clone of 'https://git.eclipse.org/r/tinydtls/org.eclipse.tinydtls' into submodule path '/opt/ESP8266_RTOS_SDK/components/coap/libcoap/ext/tinydtls' failed
#10 18.04 Failed to clone 'ext/tinydtls' a second time, aborting
```

It seems like https://git.eclipse.org/c/ has been deprecated, and the repos moved to GitHub. The HTTP redirect for the repo currently points to the github org, and not the actual repo.

ustccw, 2026-09-11:
The clone fails on the nested `ext/tinydtls` submodule inside `components/coap/libcoap`, not on ESP8266_RTOS_SDK itself.
`libcoap` still points at the old Eclipse Gerrit URL: `https://git.eclipse.org/r/tinydtls/org.eclipse.tinydtls`
That host is gone. It now HTTP-redirects to `https://github.com/eclipse-tinydtls` (the org page, not the repository), and Git will not follow that redirect.

**Workaround** — rewrite the URL before cloning or updating submodules:

```bash
git config --global url.https://github.com/eclipse-tinydtls/tinydtls.git.insteadOf https://git.eclipse.org/r/tinydtls/org.eclipse.tinydtls
```

Then:

```bash
git clone --recurse-submodules https://github.com/espressif/ESP8266_RTOS_SDK.git
```

If the SDK is already cloned:

```bash
git submodule update --init --recursive
```

Remove the rewrite later if you do not want it globally:

```bash
git config --global --unset url.https://github.com/eclipse-tinydtls/tinydtls.git.insteadOf
```

This is already fixed in the latest ESP8266_RTOS_SDK. The workaround above is only needed if you are using an older checkout such as v3.4.